kriswiner/MPU9150 #4: MPU9250 Magnetometer outputs same data in 3 axes

akhilmohan4487, 2015-05-13:
Hello,

I am trying to read magnetometer data from MPU9250 using dsPIC33F. The code was running well in MPU9150 platform and when I tried porting (with modification on the Bypass Enable bit) to MPU9250 it does not work.
I am able to read device ID and it shows 0x48. Then I did tried to read the calibration data by reading from ASAX,ASAY,ASAZ. This executes but all the data are the same. Then I tried to read the magneto registers by,

1. setting INT_PIN_CFG = 0x02 (this is as part of the initialisation of IMU)
2. power down and then setting AK8963_CNTL = 0x02 and tried 0x01 too (delay of 10ms) 
3. read magneto registers and ST2 (as in datasheet they say in continuous mode we have to read ST2)

I am able to read accelerometer and gyroscope by populating it to FIFO and this part works well.At this stage am able to read data and which is sensitive to external mag field too but the mx,my, and mz values are the same. 

Could you help me to sort out this issue ? (Think timing delay is the culprit, and I am not using Interrupt lines)
----------------------------
Thanks :)

kriswiner, 2015-05-13:
Are you doing something like this?:

void initAK8963(float * destination)
{
  // First extract the factory calibration for each magnetometer axis
  uint8_t rawData[3];  // x/y/z gyro calibration data stored here
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x00); // Power down magnetometer  
  delay(10);
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x0F); // Enter Fuse ROM access
mode
  delay(10);
  readBytes(AK8963_ADDRESS, AK8963_ASAX, 3, &rawData[0]);  // Read the x-,
y-, and z-axis calibration values
  destination[0] =  (float)(rawData[0] - 128)/256. + 1.;   // Return x-axis
sensitivity adjustment values, etc.
  destination[1] =  (float)(rawData[1] - 128)/256. + 1.;  
  destination[2] =  (float)(rawData[2] - 128)/256. + 1.; 
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x00); // Power down magnetometer  
  delay(10);
  // Configure the magnetometer for continuous read and highest resolution
  // set Mscale bit 4 to 1 (0) to enable 16 (14) bit resolution in CNTL
register,
  // and enable continuous mode data acquisition Mmode (bits [3:0]), 0010
for 8 Hz and 0110 for 100 Hz sample rates
  writeByte(AK8963_ADDRESS, AK8963_CNTL, Mscale << 4 | Mmode); // Set
magnetometer data resolution and sample ODR
  delay(10);
}

...

void readMagData(int16_t * destination)
{
  uint8_t rawData[7];  // x/y/z gyro register data, ST2 register stored
here, must read ST2 at end of data acquisition
  if(readByte(AK8963_ADDRESS, AK8963_ST1) & 0x01) { // wait for magnetometer
data ready bit to be set
  readBytes(AK8963_ADDRESS, AK8963_XOUT_L, 7, &rawData[0]);  // Read the six
raw data and ST2 registers sequentially into data array
  uint8_t c = rawData[6]; // End data read by reading ST2 register
    if(!(c & 0x08)) { // Check if magnetic sensor overflow set, if not then
report data
    destination[0] = ((int16_t)rawData[1] << 8) | rawData[0] ;  // Turn the
MSB and LSB into a signed 16-bit value
    destination[1] = ((int16_t)rawData[3] << 8) | rawData[2] ;  // Data
stored as little Endian
    destination[2] = ((int16_t)rawData[5] << 8) | rawData[4] ; 
   }
  }
}

These work for me on the Teensy and Arduino Pro Mini. The MPU9250 is
different from the MPU9150 and the differences have to be respected to get
sensible results. In particular, the MPU9150 mag, the AK8975 uses a one-shot
data read where first you have to write to a register and then read the data
every time. The 8963A on the MPU9250 has a continuous data read.

Kris

<https://github.com/kriswiner/MPU-9150/issues/4> .
<https://github.com/notifications/beacon/AGY1qs_B4gr3TyqVF7qGsQd6GTJB1I1-ks5
oIutOgaJpZM4EY0vD.gif>

akhilmohan4487, 2015-05-14:
Hello Kris,

Thanks for your reply.

I am doing the same, but still facing the same error. One thing that I noticed was I am not able to read the accelero, gyro registers directly but possible by reading the FIFO.(Don't know why) Now if I am trying to get into Fuse ROM mode, then it is affecting my acclero and gyro values. So what I did is read them before initializing the MPU module and that works.
The issue with the magnetometer is still present and I dont know how to resolve (tried to implement the method you have given). In the code provided by Invensense, the setting up of magneto is bit different and I am finding it hard to understand. 

Please let me know if you have some other way of handling this issue.

Best Regards,
Akhil

kriswiner, 2015-05-15:
Hi Akhil,

Easy solution: switch from a radio with an IMU slapped on to an IMU breakout
board with a radio slapped on. I would use a Teensy 3.1 and any MPU9250
breakout board. Try these:

https://www.tindie.com/products/onehorse/mpu9250-teensy-31-add-on-shields/

Kris 
Pesky Products

(#4)

<https://github.com/kriswiner/MPU-9150/issues/4#issuecomment-102043594> .
<https://github.com/notifications/beacon/AGY1qkT1ZuPpigpE-1YTByKmSfeDG6Lwks5
oJJ_bgaJpZM4EY0vD.gif>

akhilmohan4487, 2015-05-15:
Hello Kris,

Unfortunately I cannot change the Microchip platform as of now (dsPIC33F) and my project is getting bit delayed. Rather than moving to an another platform, I think it would be better to revert back to MPU9150 which works pretty well.
It's seems quite funny as the same set of instructions running in one and not in another. Somewhere I am making a mistake and it needs some interventions :) I will try for 2 more days and if not able to crack, will revert back to MPU9150.
Thank you very much Kris for your article and suggestions. You guys make our job easier and interesting.
Best Regards,
Akhil.

kriswiner, 2015-05-15:
Wish I could help more Akhil,

Are you sure you changed the register map correctly? The MPU92250 is not the
same as the MPU9150. If you are using the MPU9150 functions and register map
you will not get good data. Please see one of my sketches for the
differences.

Kris

(#4)

<https://github.com/kriswiner/MPU-9150/issues/4#issuecomment-102262429> .
<https://github.com/notifications/beacon/AGY1qu5JIpIW3PUjMHhaOxsatl4Vv9q2ks5
oJXhXgaJpZM4EY0vD.gif>

akhilmohan4487, 2015-05-15:
Hello,

I have changed the register map and I have cross checked now also. I am able to read the device address (both 0x71 and 0x48). Then i perform the initialization where I set gyro accelero scales, sample rate and enable the bypass mode. I am reading the register and all the values vary according to movement. But say eg. ax,ay and az are same and this is same for rest. So for reading accelero and gyro I tried FIFO and then that part is working. Magneto still give me the same bad look :)

kriswiner, 2015-05-15:
Can you simply read the raw sensor data from the accelerometer, read all six bytes and compare. the must be different. If not, then your MPU9250 board is broken (which one are you using, anyway?). If they are different, then your scaling function is somehow using the same raw data for all three axes. By reading the six accel data bytes, you can determine if this is a software or hardware error.

akhilmohan4487, 2015-05-16:
Hi,
I am able to read the accelero and gyro registers directly, but the data does not look correct and all the data are same. At the same time if I am reading via FIFO, it is correct (+/- 1g) and it varies according to my movement gyro also I am getting this way did not calibrated. What I want to know is can I use the Mastermode and use Slave 0 to read magnetometer data. Say in the FIFO R/W I can read all (accelro, gyro,temperature and magneto). I am sure first three can be done but sure magneto can also be latched in this way. What is your opinion ? 
Thanks and Regards,
Akhil.

kriswiner, 2015-05-16:
I do not believe that magnetometer data can be read into the FIFO.

Please show me your data read function. If it is correct and you are seeing all the same values in the three axis data registers of, say, the accelerometer, then your MPU9250 is broken.

It should look something like this:

void readAccelData(int16_t * destination)
{
  uint8_t rawData[6];  // x/y/z accel register data stored here
  readBytes(MPU9250_ADDRESS, ACCEL_XOUT_H, 6, &rawData[0]);  // Read the six raw data registers into data array
  destination[0] = ((int16_t)rawData[0] << 8) | rawData[1] ;  // Turn the MSB and LSB into a signed 16-bit value
  destination[1] = ((int16_t)rawData[2] << 8) | rawData[3] ;  
  destination[2] = ((int16_t)rawData[4] << 8) | rawData[5] ; 
}

akhilmohan4487, 2015-05-16:
If I am reading registers as you have mentioned in last post, the data that looks not correct.
But this function to read accelro and gyro is working (update rate = 200Hz). I am using dsPIC33FJ128GP804chip @ 20.2752MIPS input clock.

void readAccelGyroData(unsigned char MPU9250_ADDRESS,int16_t *destination){ 
// Turn OFF the temp bit in FIFO_Enable reg
    uint8_t rawData[12];
    uint16_t ii, fifo_count, packet_count;
    writeByte(MPU9250_ADDRESS, USER_CTRL, 0x40);
    writeByte(MPU9250_ADDRESS, FIFO_EN, 0x78);
    __delay_ms(25);
    writeByte(MPU9250_ADDRESS, FIFO_EN, 0x00);
    readBytes(MPU9250_ADDRESS, FIFO_COUNTH, 2, &rawData[0]); // read FIFO sample count
    fifo_count = ((int16_t)rawData[0] << 8) | rawData[1];
    packet_count = fifo_count/12;// How many sets of full gyro and accelerometer data for averaging
    for (ii = 0; ii < packet_count; ii++) {
    readBytes(MPU9250_ADDRESS, FIFO_R_W, 12, &rawData[0]);
    destination[0] = (int16_t)(((int16_t)rawData[0] << 8) | rawData[1]) ;  // Turn the MSB and LSB into a signed 16-bit value
    destination[1] = (int16_t)(((int16_t)rawData[2] << 8) | rawData[3]) ;
    destination[2] = (int16_t)(((int16_t)rawData[4] << 8) | rawData[5]) ;
    destination[3] = (int16_t)(((int16_t)rawData[6] << 8) | rawData[7]) ;  // Turn the MSB and LSB into a signed 16-bit value
    destination[4] = (int16_t)(((int16_t)rawData[8] << 8) | rawData[9]) ;
    destination[5] = (int16_t)(((int16_t)rawData[10] << 8) | rawData[11]);
    }
}

Function to read magneto

void readMagData(unsigned char MPU9250_ADDRESS,int16_t *destination)
{
// x/y/z magneto register data, ST2 register stored here, must read ST2 at end of data acquisition
  uint8_t rawData[7];
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x00);
  __delay_ms(10);
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x12);
  __delay_ms(10);
  if(readByte(AK8963_ADDRESS, AK8963_ST1) & 0x01) { // wait for magnetometer data ready bit to be set
    readBytes(AK8963_ADDRESS, AK8963_XOUT_L, 7, &rawData[0]);  // Read the six raw data and ST2 registers sequentially into data array
    uint8_t c = rawData[6]; // End data read by reading ST2 register
    if(!(c & 0x08)) { // Check if magnetic sensor overflow set, if not then report data
        destination[0] = (int16_t)(((int16_t)rawData[1] << 8) | rawData[0]);  // Turn the MSB and LSB into a signed 16-bit value
        destination[1] = (int16_t)(((int16_t)rawData[3] << 8) | rawData[2]); // Data stored as little Endian
        destination[2] = (int16_t)(((int16_t)rawData[5] << 8) | rawData[4]);
     }
  }
}

If I am not using AK8963 control, the data is zero and if used it is around 30mG for all three axes this varies due to the presence of a magnetic field.

kriswiner, 2015-05-16:
Why are you doing this:

writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x00);
__delay_ms(10);
writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x12);
__delay_ms(10);  ?

This should only be done once when you initialize the mag.

What is the length of your destination array in the main loop? Better be 6 or more!

akhilmohan4487, 2015-05-18:
Hello,
What is the length of your destination array in the main loop? Better be 6 or more! - destination array length is 3 why 6 ? (it stores mx,my and mz so it should be 3). I did not get what you meant.

kriswiner, 2015-05-18:
I meant for the accel/gyro function. If you can read the data properly from
the FIFO you should be able to read it directly from the registers. Maybe
your I2C read bytes function is broken?

(#4)

<https://github.com/kriswiner/MPU-9150/issues/4#issuecomment-103086182> .
<https://github.com/notifications/beacon/AGY1qohcdqW-g0DeME-kqZ2thJpHy-gBks5
oKfUqgaJpZM4EY0vD.gif>

akhilmohan4487, 2015-05-18:
If that is the case FIFO R/W also cannot show correct out, bit it does. accelero/gyro array length is 6. 
Thank you very much Kris for your time and effort to help me.

akhilmohan4487, 2015-05-19:
Hello,

At last magneto is working. I can directly read all the registers and it outputs good data. When I am trying to read multiple registers it is not able to do and as you suggested the error was in the I2C read bytes func. (still did not understand then how did FIFO out correct data)

Your suggestions, time everything helped me to move forward and thank you very much Kris :)

Best Regards,
Akhil

kriswiner, 2015-05-19:
Glad it's finally working for you.

akhilmohan4487, 2015-05-19:
Yea, thanks a lot.

closed 2015-05-19 by akhilmohan4487