kriswiner/MPU9150 #5: Self-test

dplbsd, 2015-05-18:
Hello, I'm trying to program the self-test for the sensor, and your code is proving to be really helpful, but so far I couldn't find the reference where it says how to get the results.

So far, the only thing I've seen is that in order to know if you pass the self-test or not, you needed the self-test value, which is defined in the datasheet as (sensor output w/out self-test - sensor output w/ self-test).

Where did you find the procedure to calculate the self-test value?
Many thanks.

kriswiner, 2015-05-18:
From Invensense, see attachment.

<https://github.com/kriswiner/MPU-9150/issues/5> .
<https://github.com/notifications/beacon/AGY1qjZvOS3uhwR1T2X1UBhbAzvNc-v8ks5
oKgjxgaJpZM4EfDdF.gif>

kriswiner, 2015-05-19:
It looks like the attachment didn't come through. Did you receive it?

dplbsd, 2015-05-19:
Nope, I didn't! I took a look at the official documentation again, but I
couldn't see anything different.

kriswiner, 2015-05-19:
It's not available in the official documentation. Send me a PM at onehorse@earthlink.net and I will send it to you.