kriswiner/MPU9150 #6: Cannot store magnetometer data into FIFO

1337hax, 2015-12-03:
Hi. Thank you for this library, it's very helpful for figuring out how to work with mpu-9150.

My current state of work with this sensor is as follows:
1. I can read 3d acc,gyro,magnetometer in single measurement mode correctly.
2. I can read 3d acc and/or gyro to FIFO correctly.
3. I can NOT read magnetometer data into FIFO. It always load zeros in places where should be magnetometer data.

I've tried many approaches, example approach is here (it's based on your code):
```
 mpu9150a_driver_write_register(this, MPU9150_RA_INT_PIN_CFG, 0x02); // enable bypass to have access to magnetometer 
  // Configure Magnetometer for FIFO
  // Initialize AK8975A for write
  mpu9150a_driver_write_register(this, MPU9150_RA_I2C_SLV1_ADDR, 0x0C);  // Write address of AK8975A
  mpu9150a_driver_write_register(this, MPU9150_RA_I2C_SLV1_REG, 0x0A);   // Register from within the AK8975 to which to write
  mpu9150a_driver_write_register(this, MPU9150_RA_I2C_SLV1_DO, 0x01);    // Register that holds output data written into Slave 1 when in write mode
  mpu9150a_driver_write_register(this, MPU9150_RA_I2C_SLV1_CTRL, 0x81);  // Enable Slave 1
  // Set up auxilliary communication with AK8975A for FIFO read
  mpu9150a_driver_write_register(this, MPU9150_RA_I2C_SLV0_ADDR, 0x8C); // Enable and read address (0x0C) of the AK8975A
  mpu9150a_driver_write_register(this, MPU9150_RA_I2C_SLV0_REG, 0x03);  // Register within AK8975A from which to start data read
  mpu9150a_driver_write_register(this, MPU9150_RA_I2C_SLV0_CTRL, 0x86); // Read six bytes and swap bytes
  
  mpu9150a_driver_write_register(this, MPU9150_RA_FIFO_EN, 0x79); // capture gyro and accelerometer and slave 0 (magnetometer) to fifo
  mpu9150a_driver_write_register(this, MPU9150_RA_I2C_MST_DELAY_CTRL, 0x80); // Enable delay of external sensor data until all data registers have been read
  
  mpu9150a_driver_write_register(this, MPU9150_RA_USER_CTRL, 0x40);   // Enable FIFO  
```
Raw data in my fifo looks like this:
```
-1851 | -983 | -3746 // acc
-83 | 8 | -32 // gyro
0 | 0 | 0 // magn
-1858 | -997 | -3720 // acc
-54 | 4 | -17 // gyro
0 | 0 | 0 // magn
-1883 | -997 | -3745
-56 | 5 | -19
0 | 0 | 0
-1823 | -985 | -3718
-57 | 4 | -19
0 | 0 | 0
-1857 | -998 | -3764
-54 | 5 | -20
0 | 0 | 0
-1876 | -997 | -3751
-55 | 5 | -17
0 | 0 | 0
-1860 | -993 | -3747
-54 | 6 | -19
0 | 0 | 0
-1830 | -981 | -3746
-52 | 4 | -19
0 | 0 | 0
-1846 | -981 | -3764
-57 | 4 | -20
0 | 0 | 0
-1837 | -1002 | -3765
-53 | 4 | -20
0 | 0 | 0
```

Any ideas?

kriswiner, 2015-12-03:
As far as I know, the AK8963C mag data can not be loaded into the FIFO. I am not sure why this is so, since other slave data can be, but if you look at Invensense's own SDK there are numerous comments in the code to this effect and it is probably one reason why the DMP cannot do 9 DoF sensor fusion.

You could try to use an external magnetometer and see if you can get this data into the FIFO, but the embedded mag data won't.

1337hax, 2015-12-03:
OK, I see. Thank you for quick response.

closed 2015-12-03 by 1337hax