kriswiner/MPU9150 #7: mRes is wrong?

yang614, 2016-07-14:
Hi,Kriswiner 

The mRes = 10.*1229./4096. is 3 , the comment says  // Conversion from 1229 microTesla full scale (4096) to 12.29 Gauss full scale //actually  has no change .

I read the datasheet ,I think the mRes should be 1229./4096. not 10.*1229./4096.  or tell me where I have the misunderstanding.

kriswiner, 2016-07-14:
The 10 converts microTesla to milliGauss, which I prefer.

teks5qVdU_gaJpZM4JMHgU> .
<https://github.com/notifications/beacon/AGY1qjGDjM4bZf6f-t271vGZnaTLEmwsks5
qVdU_gaJpZM4JMHgU.gif>

yang614, 2016-07-14:
Oh,I see 
I have another question is why set the gyro &  acc FS_SEL 0 and divide the raw data by 4&8 instead of 
setting to 3 directly?

yang614, 2016-07-14:
Ok, I see .
and I have another question where is the " conform to expected bias input format" in  calibrateMPU9150(float * dest1, float * dest2) ?? I cant find it from the datasheet.

kriswiner, 2016-07-14:
Consult the data sheet.

Poks5qVeSFgaJpZM4JMHgU> .
<https://github.com/notifications/beacon/AGY1qr-k9vI2PNON_NyVeyaqtnErEwdcks5
qVeSFgaJpZM4JMHgU.gif>

kriswiner, 2016-07-14:
There should be some discussion of the bias registers in the data sheet. In
any case IIRC the data format is expected to be counts at 8 g full scale.

aiks5qVenrgaJpZM4JMHgU> .
<https://github.com/notifications/beacon/AGY1qtJdGQTYAInRYD3FWcC4PB5T2Fmmks5
qVenrgaJpZM4JMHgU.gif>