kriswiner/MPU9150 #8: how to access the magnetometer

jenextech, 2017-03-02:
hello kriswiner,
    actually i have used mpu9250 library and i have got reading of the accelerometer and gyrometer data,but when i tried to communicate with mpu9250,i could not.
 Means that after reading data of accel. and gyro ,i have enable the bypass_i2c.then when try to something write/read with magnetometer_address(0x0C) and send start_condition of i2c ,it could not get acknowledgement.

kriswiner, 2017-03-02:
If you perfrom an I2C scan, which devices show on the I2C bus after enabling bypass?

jenextech, 2017-03-06:
thank you.
actually because of some hardware issue,it cause communication problem with the magnetometer.
now,i can get all data,see below. are all right?

ax             ay           az           (meter/second square)
-0.02      -0.01       -0.32

gx             gy          gz           (degrees/second)
-0.33      29.65      -0.00

mx            my           mz        (milliGauss)
-461.03   -275.13    5563.25

temperature=23.19 degrees centigrade

kriswiner, 2017-03-06:
The mag values seem about 10x too large, and the gyro values are very large
indeed if the device is at rest.

jenextech, 2017-03-08:
okay.so,what should i change?..
any specific setting or anything..
and i have 1 question that how can i sure that raw data is right?..any hint?..

kriswiner, 2017-03-08:
check units