kriswiner/MPU9150 #10: Accelerometer calibration

robopassio, 2018-09-11:
I use this source with an Arduino Uno board, I have a problem with the calibration of the accelerometer. If the initial posture of the board is leaned then there is an offset and it can not get back to the true value after calibrating. I tried to remove the following lines, then the offset dissapeared but I do not know how to calibrate the accelerometer in a right way.
//  // Push accelerometer biases to hardware registers
//  writeByte(MPU9150_ADDRESS, XA_OFFSET_H, data[0]);
//  writeByte(MPU9150_ADDRESS, XA_OFFSET_L_TC, data[1]);
//  writeByte(MPU9150_ADDRESS, YA_OFFSET_H, data[2]);
//  writeByte(MPU9150_ADDRESS, YA_OFFSET_L_TC, data[3]);
//  writeByte(MPU9150_ADDRESS, ZA_OFFSET_H, data[4]);
//  writeByte(MPU9150_ADDRESS, ZA_OFFSET_L_TC, data[5]);

Please give me some ideas about this.

kriswiner, 2018-09-12:
The accel has to be calibrate when lying flat and motionless.

robopassio, 2018-09-12:
It is left in an inclined surface (without any motion), then I realize that there was an offset.
The angle offset is reduced when the inclined angle is reduced, so I cannot trust the output angle.
The output is more reasonable when I commented on the above lines. Can you test it again or give me some instructions?
Thanks.

kriswiner, 2018-09-12:
Calibrate the device when laying flat and motionless.

robopassio, 2018-09-12:
Of course, it is fixed in the surface, but a smaller inclined surface will produce a smaller offset in the angle.

kriswiner, 2018-09-12:
I do not understand what you are asking me then...

kriswiner, 2018-09-12:
> I mean that I doubt on the calibration of the accelerometer. Why did the
> output angles deviate an amount (constant offsets) from its true value when
> I fixed it on a small inclined surface.
> I think that the calibration is always true even in an inclined surface as
> long as it is motionless. However, it seems to depend on the initial lean
> angle of the surface that the sensor lied on. This is unreasonable because,
> in some application, the sensor is set up on an arbitrary surface.

Agreed, it should read 0, 0, 1 if you calibrate even at an incline. Sounds
like you are not calibrating correctly.

robopassio, 2018-09-12:
Because the offset of the angles is varied depending on the inclined angle of the surface. So I just think that even I put in a slightly inclined surface, the output angles from my sensor still not correct, because it still exists a small offset. Do you agree with me at this point? If not please explain or give me a solution for this.

kriswiner, 2018-09-12:
I think if you properly calibrate with the sensor even at an angle you will
see 0, 0, 1 output from the accel. if not, there is something wrong with
otur calibration function.

Civilduino, 2019-01-07:
Hello, it is the same question I asked in the MPU-6050 repository. This is not a "real" calibration but an offset correction. This method assumes that the surface where the sensor lies is perfectly aligned with the gravity vector and therefore the offset from the readings are subtracted to match the "reality".

However, it is extremely difficult to have an ideal surface aligned to the gravity vector. So the true angle with respect to gravity is hard to obtain unless you have expensive equipment or do a more proper calibration. See the MPU-6050 question.

You are just correcting the angles based on the surface you are using.

kriswiner, 2019-01-07:
If you want absolute orientation you will need more than just an
accelerometer and gyro, you need a magnetometer too, all three need to be
corrected for offset biases, but then the angle wrt any other vector is
unique, absolute, and correct to withing ~2 degrees or so depending on
calibration.

Yes, if you just want to know the ange wrt gravity, you can use an
accelerometer and the accuracy will be dependent on how well you remove the
offset biases and how well the alignment is to gravity when you do so.

So what?