kriswiner/MPU9150 #11: intPin useless?

stackmagic, 2019-08-17:
is the "intPin" used at all? it is set as an input but that's all i can find - this variable could be deleted I guess?
https://github.com/kriswiner/MPU9150/blob/697032ac72af5795db2f4b64f9ccc27710afc796/MPU9150BasicAHRS.ino#L211

kriswiner, 2019-08-17:
It's up to you. The sketch is not meant to be a black box but an example of
what could be done. It is likely in the sketch the data ready registers are
being read (polling) instead of using the interrupt. Latter is preferred.

stackmagic, 2019-08-20:
Yes it look like polling is used because I only find the declaration of "intPin" and its pinMode() setup, nothing else... that's why I asked in the first place... so maybe that's just a remainder from an older version?

closed 2019-08-20 by stackmagic

kriswiner, 2019-08-20:
It's easy enough to modify the sketch to make use of the interrupt. Not
sure what the question is...